freifunkh/site #4: More site adjustments

lemoer, 2017-03-05:
Hi,
some recently merged changes in the Gluon master require adjustments to the
site configuration:

* The package gluon-next-node has been removed, it has been merged into
gluon-client-bridge and gluon-mesh-vpn-batman-adv-*. gluon-next-node needs
to be removed from site.mk
* New i18n keys "gluon-config-mode:altitude-label" and
"gluon-config-mode:altitude-help": As different communities have different
ideas how to use this field, it seemed useful to make the label and help
text configurable. You can find the old texts in docs/site-example of the
Gluon repo.

-- NeoRaider

AiyionPrime, 2017-03-07:
Currently we're not using the altitude at all,
not that I would not like the idea, but we've no use for the feature for now.
So for now I'd say we don't change that, but keep in mind, that we would have the ability to.

The other bulletpoint is quite trivial, I'll have a look at it in a few minutes.

lemoer, 2017-03-10:
Ah I didn't know that this changes are already included in `2016.2.3`. I thought they were only in the master branch...

lemoer, 2017-03-10:
After talking to @AiyionPrime, it turns out that he only used this for later merging. Merging this does not make any sense, until we build from the master branch. So I'll reopen this issue.

lemoer, 2017-03-10:
I'll add the blocked label to mark that it should be done after #12

lemoer, 2017-03-16:
I'll close this issue as this is applied in the lede branch as c8074a1c65a6e45d73ec67d240c5af5dd5bc2d0d. It will be merged with #12 into the master branch.

closed 2017-03-16 by lemoer

lemoer, 2017-03-16:
Sorry... Push-forcing was again needed :/ So the commit is now applied as 79bd34e1a6d3ff6afad1f8ab95d43e9d97873159

PR #5 (2017-03-07): removed gluon-next-node
This removes gloun-next-node like we talked about in issue #4 .
Please review as this closes #4
commit cabb9d7f (2017-03-07): removed gluon-next-node

This removes gloun-next-node like we talked about in issue #4 .
Please review as this closes #4