freifunkh/site #10: build scripts

AiyionPrime, 2017-03-10:
It may be a nice feature to have build scripts for our build instances.
Like having only one shell file which gets executed by our jenkins-servers and does all the neccessary work.

lemoer, 2017-03-10:
I would propose logging to the following path:
``` shell
logs/$(date --rfc-3339=seconds | tr " " "T")_${GLUON_TARGET}
```

lemoer, 2017-03-10:
Maybe we could integrate #8

lemoer, 2018-08-24:
We are now running the jenkins server for quite a while, while also providing nightly builds with it. Do you have any objections whether we should store the build procedure here?

My objections are:
- I often modify the build procedure to do some patching before the actual build, when I want to test something. This wouldn't be so easy anymore and would cause a lot of messy git commits.
- Updating the build procedure in general becomes more tedious. The scripts now pull the site repo, but if they are located in the site itself, then we have a chicken egg problem.

I would vote for closing with wontfix. Does anyone else have oppinions?

AiyionPrime, 2018-08-24:
I hoped that by now we would've moved towards something like declarative pipelines; without pushing it.
I've got to admit, though I don't like having our builds running like now without jenkinsfiles is not my favourite, but doing got enough.
I'd not care about messy commits; we could store the repo in private if you care.
But I'd not use Jenkinsfiles just to use Jenkinsfiles; but because theyre superior to what we're doing now.
As long as they aren't, we shouldn't use them at all :/
Therefore vote for wontfix, for now.

1977er, 2018-08-24:
Maybe we can manually copy/paste the build scripts to a repo. I like the idea of having a well defined place to restore from after a crash.

lemoer, 2018-08-24:
> I like the idea of having a well defined place to restore from after a crash.

Me aswell. Should we keep this issue open until this is done?

CodeFetch, 2021-02-22:
Totally outdated. Jenkins works great now.

closed 2021-02-22 by CodeFetch