freifunkh/site #8: broken as experimental

AiyionPrime, 2017-03-10:
We came to the conclusion it might be nice to build broken targets as experimental every once in a while.

1977er what do you think about this?

lemoer, 2017-03-10:
Do you think it would be helpful if we have a `BUILD_ONLY_BROKEN_TARGETS` flag in the gluon build scripts?

AiyionPrime, 2017-03-10:
Didn't know that was a thing.
Yes i like the idea. We could reduce build time by deploying the stable unbroken targets in the experimental folder as well.

lemoer, 2017-03-10:
Maybe we should introduce another "firmware branch" for this. I think experimental should be available for testing stuff like `ddhcpd` or #7

lemoer, 2017-03-10:
Maybe a branch called "broken"?

AiyionPrime, 2017-03-16:
I'd leave this here for another 16 hours, then create a new branch.
How about we have a look at how much effort this costs midterm;
and decide whether we keep it on the 13th april?

1977er, 2017-03-16:
Once we have a distributed build environment, build extra stuff like this may be a useful utilisation.

tobby88, 2017-03-16:
Would be great if the more stable channels are always released in the less stable channels as well.
e.g. if a new beta is released, it should be flashed to all devices on experimental channel as well and if a new stable is released, it should be flashed to all devices on experimental and beta, too.

Concerning the original question: Yes, build broken targets once in a while sounds good, since many of these aren't really broken, but just not tested.

AiyionPrime, 2017-03-19:
No, that would not always be great.
While we're testing e.g. new mesh-modules we do not want the test beeing interrupted by gluon-18d.

With beta this would mostly be fine, but experimental should not even neccessary be a channel, where routers stay for long.
In my opinion it would be more useful if users had to opt in for a specifc experiment. (And had in return a better overview over their devices performance.)