freifunkh/site #13: Test package gluon-ebtables-segment-mld

lemoer, 2017-03-14:
[Link](https://gluon.readthedocs.io/en/v2016.2/package/gluon-ebtables-segment-mld.html)

lemoer, 2017-03-14:
This may reduce the "groundnoise"

lemoer, 2018-01-04:
We should really consider use this, as a a lot of our groundnoise is actually mld.

1977er, 2018-01-04:
You have my blessings. ;)

AiyionPrime, 2018-01-31:
Das mrd stark abwesend ist, ist us egal?

lemoer, 2018-01-31:
@AiyionPrime could you elaborate?

AiyionPrime, 2018-02-01:
Possible that we simply do not care;
but in case someone of our 2000 clients is using his freifunk-node in a more sophisticated manner than just in order to provide a guest wifi; we should at least consider writing a memo onto the mailing list that we'll let the routers drop certain mld messages.

Using the package may lead snooping switches and routers attached to possibly miss, that the connected freifunk node is a multicast router and may therefore fall back to eg. unicast messages to a bunch of possible recipients in the freifunk-network;
I'm not sure wether someones actually streaming something into the net, but see no reason why not.

Maybe you're right and its unlikely that someones using this.

lemoer, 2018-02-19:
It would be nice to measure the effect on the groundnoise. So it would be nice, if we begin to measure the groundnoise min ~1week before we start to distribute this update in stable branch.

lemoer, 2018-03-23:
It's no own package any more since fe186077714a346da751daef79d238dab95c6263 and upstream https://github.com/freifunk-gluon/gluon/commit/17370d8462637a7d53bc1705e49e542acd92455e.

closed 2018-03-23 by lemoer