freifunkh/site #14: Drop openwrt/lede opkg mirrors

lemoer, 2017-03-16:
This is first possible, if we have public IPv6 in our network.

lemoer, 2018-08-24:
This is no longer blocked as we have now public ipv6.

We are talking about this line in the `site.conf` and about replacing it by the direct upstream url. This would enable us to to shutdown the proxy:

``` lua
lede = 'http://opkg.ffh.zone/lede/releases/packages-%v/%A',
```

1977er, 2018-08-24:
Go for it.

But for legacy nodes we should not shutdown the proxy right away.

lemoer, 2018-08-24:
At least we can held it online for a while, but this is just for package installing via opkg. I think it doesn't really matters for outdated routers as nobody seems to maintain them anyway, so also nobody will try to install packages.

1977er, 2018-08-24:
true.

Dark4MD, 2020-01-25:
Done in 7a24e1e1711e172e4515a5d29fb9f66d57480b98

closed 2020-01-25 by Dark4MD