freifunkh/site #16: Use respondd-module-airtime package

lemoer, 2017-07-08:
In the master branch, there is a new package `respondd-module-airtime` collecting information about the wifi airtime. We could use this as soon, as we are switching to `2017.x`

[Link to the module information](https://github.com/freifunk-gluon/packages/tree/master/net/respondd-module-airtime)

lemoer, 2018-01-07:
Will be built in next nightly build.

lemoer, 2018-02-15:
Can be tested using ```gluon-neighbour-info -d ::1 -p 1001 -r statistics``` on a node.

The output should be like:

```
{
  "statistics": {
    "wireless": [
      {
        "frequency": 5220,
        "active": 366561161,
        "busy": 46496566,
        "rx": 808415,
        "tx": 41711344,
        "noise": 162
      },
      {
        "frequency": 2437,
        "active": 366649704,
        "busy": 205221222,
        "rx": 108121446,
        "tx": 85453679,
        "noise": 161
      }
    ]
  }
}
```

Can someone test this?

AiyionPrime, 2018-02-15:
Will do, tomorrow.

lemoer, 2018-02-16:
Seems to work!

lemoer, 2018-02-16:
But we have no grafana or other tool, which shows it.

1977er, 2018-02-18:
Yo, tut.

```javascript
{"wireless":[{"frequency":5220,"noise":164,"active":250976,"busy":4063,"rx":98,"tx":1725},{"frequency":2412,"noise":161,"active":250480,"busy":119823,"rx":96882,"tx":16934}],"clients":{"total":1,"wifi":1,"wifi24":0,"wifi5":1},"traffic":{"rx":{"packets":51987,"bytes":4148372},"tx":{"packets":233,"bytes":26230,"dropped":20},"forward":{"packets":0,"bytes":0},"mgmt_rx":{"packets":11882,"bytes":4333796},"mgmt_tx":{"packets":19710,"bytes":8120544}},"gateway":"88:e6:40:20:80:01","gateway_nexthop":"22:5a:4c:a7:ec:b1","mesh_vpn":{"groups":{"backbone":{"peers":{"sn03":null,"sn02":null,"sn01":null,"sn10":null,"sn04":null,"sn09":null,"sn06":null,"sn07":{"established":242.510},"sn08":null,"sn05":null}}}},"gateway6":"d6:62:9d:f2:91:ea","node_id":"e894f63f3bf8","time":1518957657,"rootfs_usage":0.0689,"memory":{"total":125240,"free":89324,"buffers":2296,"cached":7996},"uptime":282.81,"idletime":169.24,"loadavg":0.87,"processes":{"running":2,"total":68}}
```

lemoer, 2018-05-12:
Works for me.

closed 2018-05-12 by lemoer