freifunkh/site #18: Consider use of gluon-web-mesh-vpn-fastd package

lemoer, 2017-08-25:
`
gluon-web module to enable and disable encryption for the mesh VPN
`

- This is only useful, if we also enable it also on gateway side
- Note that the package name was `gluon-luci-mesh-vpn-fastd` before lede.

lemoer, 2018-01-07:
1. It's already enabled on gateway side.
2. This is an optional feature in expert mode, so I don't see any strong oppions in here.

I think, after testing we should announce it our release notes.

lemoer, 2018-01-07:
Will be available in the next nightly builds.

lemoer, 2018-02-15:
[image]

1977er, 2018-02-15:
Mal schauen, ob das wirklich "Geschwindigkeitsvorteile" bringt. Im aktuellen Netz tut es das eher nicht.

lemoer, 2018-02-15:
Works. I'll close it.

closed 2018-02-15 by lemoer