freifunkh/site #20: Drop 841* in Download section on the Website.

lemoer, 2017-08-30:
I think we should still provide upgrades and provide the image when somone asks per email.

lemoer, 2017-08-30:
Well... Maybe we should directly remove all 4MB devices.

lemoer, 2017-08-30:
@Suuush @1977er

1977er, 2017-08-31:
123 von 147 Images sind kleiner als 4M.

lemoer, 2017-08-31:
@1977er Oh. Okay. Find ich jetzt aber noch nicht, dass das ein Hindernis ist?

1977er, 2017-08-31:
Ich weiß nicht, ob man von der Imagegröße auf den verbauten Speicherstein setzen kann, aber ggf. flögen dann mehr Plattformen raus. Sieht vielleicht auf dem Blatt Papier nicht mehr so schön aus. Auch, wenn es in der Praxis bisher keine Rolle gespielt hat, ob wir Exotenchipsatz XYZ unerstützt haben oder nicht.

lemoer, 2017-08-31:
> Ich weiß nicht, ob man von der Imagegröße auf den verbauten Speicherstein setzen kann, aber ggf. flögen dann mehr Plattformen raus.
Du meinst, es gibt Router, die in der gleichen Hardware-Revision mit unterschiedlichen Flash-Größen verkauft werden?

> Sieht vielleicht auf dem Blatt Papier nicht mehr so schön aus. 
Du meinst, dass wir so wenig Router unterstützen?

> Auch, wenn es in der Praxis bisher keine Rolle gespielt hat, ob wir Exotenchipsatz XYZ unerstützt haben oder nicht.
Die Aussage habe ich nicht verstanden.

1977er, 2017-08-31:
s/setzen/schließen/

Zum Beispiel gibt es den 710N in einer Hardwarerevision mit 8MB, bei der das Image aber nur 4MB groß ist. Ich weiß nicht, wie viele andere Images das noch betrifft. Vielleicht sind viele der aktuell unterstützten Plattformen doch schon 8MB Geräte oder größer.

Ich fände es optisch unschön, wenn wir dann nur noch eine Handvoll Geräte unterstützen würden. - Auch, wenn es dann die Geräte sind, die eh nur nachgefragt werden. Geht mir hier nur um den kosmetischen Aspekt, große Zahlen bei der Anzahl unterstützter Plattformen liefern zu können. ;-)

AiyionPrime, 2017-09-10:
Ich würde erstmal nur die 841er aus dem Downloadbereich entfernen,
dass sind die einzigen Geräte, die uns quantitativ in Probleme reiten werden.
Ansonsten möchte ich festhalten, dass wir vorerst ja nur 'nicht mehr empfehlen' und nicht 'nicht mehr unterstützen' was die ollen Krücken angeht.

AiyionPrime, 2018-08-24:
Weißt du, warum die ubiquiti und den vocore eol gesetzt haben?

1977er, 2018-08-24:
> Darüber hinaus haben wir beobachtet, dass Geräte mit 8MB Flash-Speicher besonders betroffen sind, bei uns sind das vor allem Geräte von Ubiquiti sowie der TL-WR1043ND v1 von TP-Link.

CodeFetch, 2019-06-21:
Maybe this issue can be closed as the next Gluon version will deactivate generation of factory images for these devices by default?

Dark4MD, 2020-02-04:
Es werden seit vH18 für beta und stable keine Factory Images mehr für legacy Geräte (4/32) gebaut. Upgrades, auch von einem mit OpenWRT geflashten Router, sind weiterhin möglich.

closed 2020-02-04 by Dark4MD