freifunkh/site #21: Migrate all x86 nodes (esp. offloader)

1977er, 2017-09-29:
LEDE firmware images need at least 2016.2.7 as a middle step in order to maintain config. Autoupdater is often deactivated on virtual machines or x86 instances.

Each node needs special TLC.

lemoer, 2018-02-15:
Deadline is 1st march 2018.

lemoer, 2018-02-15:
> Liebe Freifunker,
> 
> einige von euch betreiben Freifunk-Knoten auf x86-Geräten (also
> PC-Hardware). Darunter fallen unter anderem qemu, virtualbox, vmware,
> ... & co.
> 
> Da sich die Firmware um einen Sprung vergrößert hat, reicht die
> Partitionsgröße der bisher angebotenen Images für die Zukunft nicht mehr
> aus. Die neuen Firmwares benötigen mindestens 273 MB Speicher.
> 
> Für die Betreiber gibt es zwei Möglichkeiten:
> 
> a) Eine Vergrößerung des bestehenden Images. Dadurch spart man sich das
>    Neueinrichten. Für qemu funktioniert das folgend:
> 
> qemu-img resize $IMAGE 273MB
> 
> b) Ein neues Image benutzen. Da wir gerade noch keine Beta releast
>    haben, steht hier der nightly-build zur Verfügung:
> 
> https://build.ffh.zone/job/gluon-nightly/ws/download/images/factory/
> 
> Demnächst wird der nightly-build in ein Beta-Image gefreezt und dann
> irgendwann auch als stable angeboten. Ansonsten sollte der
> nightly-Branch relativ stabil sein. Falls jemand Fehler bemerkt, so ist
> er/sie herzlich dazu aufgerufen, diese via IRC oder auf der ML zu melden.
> 
> Als Frist setzen wir den **01. März 2018**. x86-Geräte, die danach nicht
> entsprechend resized sind, werden beim nächsten Firmware-Release kaputt
> gehen, da der Autoupdater die Partitionsgrößen nicht prüft und somit ein
> "halbes Image" schreiben wird und danach crashen wird.
> 
> Gruß,
> Leo

1977er, 2018-02-18:
Actually just all virtual x86 instances need TLC. I tested a Futro and the autoupdater automatically resized the partition without special treatment.

lemoer, 2018-03-23:
Time is over. Closing.

closed 2018-03-23 by lemoer