freifunkh/site #30: Wrong labels in domain selector

moridius, 2018-04-22:
The domain selector in config mode doesn't show pretty labels yet.

[image]

lemoer, 2018-04-29:
Not sure yet, why this is a problem yet?

German proposal from @bschelm:
> "Hier kannst du deine Standortzugehörigkeit festlegen. Wähle deinen Stadtteil oder deine Stadt aus. Passt keine Auswahl, lass es einfach bei Default stehen. Die Standortzugehörigkeit legt fest, mit welchen Freifunk Knoten sich dein Knoten verbinden kannst (Mesh-Netz)."

lemoer, 2018-05-12:
TODO:
- [ ] Es sollte keinen default geben.

lemoer, 2018-05-12:
Should be fixed in next nightly.

lemoer, 2018-05-13:
[image]

closed 2018-05-13 by lemoer