freifunkh/site #37: autoupdater-wifi-fallback is broken

lemoer, 2019-02-22:
In nightly the *autoupdater-wifi-fallback* pakage is broken:

```
gluon-reconfigure
[...]
Configuring: 510-autoupdater-wifi-fallback
/usr/bin/lua: ./510-autoupdater-wifi-fallback:3: module 'nixio.fs' not found:
        no field package.preload['nixio.fs']
        no file './nixio/fs.lua'
        no file '/usr/share/lua/nixio/fs.lua'
        no file '/usr/share/lua/nixio/fs/init.lua'
        no file '/usr/lib/lua/nixio/fs.lua'
        no file '/usr/lib/lua/nixio/fs/init.lua'
        no file './nixio/fs.so'
        no file '/usr/lib/lua/nixio/fs.so'
        no file '/usr/lib/lua/loadall.so'
        no file './nixio.so'
        no file '/usr/lib/lua/nixio.so'
        no file '/usr/lib/lua/loadall.so'
stack traceback:
        [C]: in function 'require'
        ./510-autoupdater-wifi-fallback:3: in main chunk
        [C]: ?
[...]
One or more upgrade scripts failed. Please review the above error messages.
root@noise-19:~# 
```

I think this is because of switch from `nixio` to `luaposix` in gluon.

lemoer, 2019-02-22:
I am not sure, whether the release candidate v13 is also affected?

AiyionPrime, 2019-03-01:
I am sure, it does not on your device:
```
BusyBox v1.28.4 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06-SNAPSHOT, r7629+12-eef6bd3393
 -----------------------------------------------------
root@crashtester:~# gluon-reconfigure 
Configuring: 001-upgrade
Configuring: 005-site-domain
Configuring: 010-primary-mac
Configuring: 020-interfaces
Configuring: 030-system
Configuring: 100-core-reset-sysctl
Configuring: 100-lock-password
Configuring: 110-network
Configuring: 120-ntp-servers
Configuring: 140-firewall-rules
Configuring: 150-poe-passthrough
Configuring: 200-wireless
Configuring: 210-interface-wan
Configuring: 220-interface-lan
Configuring: 300-gluon-client-bridge-network
Configuring: 300-setup-mode
Configuring: 310-gluon-client-bridge-local-node
Configuring: 310-gluon-mesh-batman-adv-mesh
Configuring: 310-setup-mode-migrate
Configuring: 320-gluon-client-bridge-wireless
Configuring: 320-gluon-mesh-batman-adv-client-bridge
Configuring: 320-setup-ifname
Configuring: 330-gluon-mesh-batman-adv-mac-addresses
Configuring: 400-mesh-vpn-fastd
Configuring: 400-neighbour-info-firewall
Configuring: 400-respondd-firewall
Configuring: 410-mesh-vpn-fastd-generate-secret
Configuring: 500-autoupdater
Configuring: 500-mesh-vpn
Configuring: 500-node-info-system
Configuring: 500-opkg
Configuring: 500-radvd-remove-user
Configuring: 500-status-page
Configuring: 510-autoupdater-wifi-fallback
Configuring: 510-node-info-role
Configuring: 520-node-info-whitespace-fix
Configuring: 800-migrate-batadv
Configuring: 820-dns-config
Configuring: 998-commit
Configuring: 999-version
root@crashtester:~# cat /lib/gluon/release
vH13
root@crashtester:~# 
```

lemoer, 2019-03-01:
IIRC you updated the feed for the wifi fallback updater. The issue seems to be resolved because of that.

closed 2019-03-01 by lemoer