freifunkh/site #46: veraltete Versionen und Drachen

AiyionPrime, 2020-04-14:
Bernd hat das zuletzt im Januar angemerkt, wir haben wie immer einen statlichen Anteil an veralteter Firmware, die sich nicht mehr von alleine Erholen wird.

Wir haben übrigens immer noch Geräte mit <vH17, die autoupdater auf stable haben und trotzdem nicht updaten. Da sollte noch mal etwas drauf geachtet werden beim nächsten Update.

_Originally posted by @bschelm in https://github.com/freifunkh/site/pull/43#issuecomment-572475922_

AiyionPrime, 2020-04-14:
Das hier ist die benannte Liste. Wer seinen Router hier findet, sollte Ihn aktualisieren.
Am saubersten ist es mMn., die Router neu aufzusetzen und die alte Konfiguration nicht zu behalten.

AUE-STB28-Garten
DieFallinger10
Kasparland1
Am-Hagen-26-7520
UR
Schwuebl.2B_II-002719fc8b66
Schmarrie-Mitte
IdeenArena-STH_3
TauschTreff Hannover
NDS-Kafkastr-Haltestelle-Uplink
Freifunk-LT21
freifunk-a0f3c1d8b7ee
ffh-pattensen-bw-me
AhBaDaStadtbezirksrat02/1
StadtRinteln11
freifunk-f4f26d53de5e
dir842c1-test
sgs-MB-WEB
Test
NDS-Summersville_Block
Marienau-Knickstraße1
sharky-alpha
BBS-Neustadt-a-Rbge_ZSH
NDS-The-Doctor
FFS-E Freibad
FFH-CFIT-LIT
freifunk-fworld-exp
freifunk-50c7bf14b254
NDS-Kathrin-Oliver
Agenda21Haus72-841-ext
Freden-hats-Bahnhof
Freifunk-LT52
St.Petri-Steinwedel
pds5-jeremy
Holodeck-2
freifunk-Schneeren4
freifunk-HofGrindau
freifunk-50c7bf06b7ff
ffh-Davenstedt-Nord

lemoer, 2020-04-15:
Wenn ich deine Liste genauer untersuche, dann teilt sich das so:

Offline:
```
AUE-STB28-Garten
DieFallinger10
UR
Schwuebl.2B_II-002719fc8b66
IdeenArena-STH_3
freifunk-f4f26d53de5e
sgs-MB-WEB
NDS-Summersville_Block
Marienau-Knickstraße1
NDS-The-Doctor
FFH-CFIT-LIT
freifunk-50c7bf14b254
Agenda21Haus72-841-ext
St.Petri-Steinwedel
pds5-jeremy
freifunk-HofGrindau
```

Autoupdater deaktiviert:
```
NDS-Kafkastr-Haltestelle-Uplink: False stable
AhBaDaStadtbezirksrat02/1: False stable
sharky-alpha: False nightly
freifunk-fworld-exp: False experimental
Freden-hats-Bahnhof: False stable
Freifunk-LT52: False stable
Holodeck-2: False stable
freifunk-50c7bf06b7ff: False nightly
```

Autoupdater aktiviert:
```
Kasparland1: True stable
Am-Hagen-26-7520: True labor
Schmarrie-Mitte: True nightly
TauschTreff Hannover: True stable
Freifunk-LT21: True stable
freifunk-a0f3c1d8b7ee: True stable
ffh-pattensen-bw-me: True stable
dir842c1-test: True stable
Test: True True stable stable
BBS-Neustadt-a-Rbge_ZSH: True beta
FFS-E Freibad: True stable
NDS-Kathrin-Oliver: True stable
freifunk-Schneeren4: True stable
ffh-Davenstedt-Nord: True stable
```

lemoer, 2020-04-15:
Für die mit aktiviertem Autoupdater habe ich hier mal die Firmwareversion herausgefunden:
```
Kasparland1: vH19
Am-Hagen-26-7520: Manu-Labor.2020-03-16
Schmarrie-Mitte: vH0.20200325
TauschTreff Hannover: WR841N-MOD-v2018.2.1
Freifunk-LT21: vH10
freifunk-a0f3c1d8b7ee: hannover-3
ffh-pattensen-bw-me: vH14
dir842c1-test: ManuWTF-LaborWin
Test: WR841N-MOD-v2018.2 vH18
BBS-Neustadt-a-Rbge_ZSH: Manu-Labor.2020-01-31
FFS-E Freibad: vHN20180703
NDS-Kathrin-Oliver: vH12
freifunk-Schneeren4: vH10
ffh-Davenstedt-Nord: vHN20180603
```

AiyionPrime, 2020-04-16:
Mein Fehler. Ich habe die Auswertung mit nodestk schon gemacht. Ich wollte hier eigentlich nur anstoßen, dass Ihr eure Router schubst und wie wir weiter verfahren wollen.

Dass Router offline sind, spielt für Ihre Klassifizierung mMn. keine große Rolle, weil sie erwartungsgemäß zeitnah wieder da sind. (Sie waren schließlich auch in den letzten drei Wochen irgendwann da.)

AiyionPrime, 2023-03-24:
Hier hat sich einiges getan. Die Geräte, die älter als vH25 sind, sind jetzt nur noch fünf an der Zahl:

- [Niki-restaurant.de_Das_Griechische_Restaurant](https://hannover.freifunk.net/karte/#/en/map/60e327be1740)
- [Holodeck-2](https://hannover.freifunk.net/karte/#/30b5c222d872)
- [LTr2](https://hannover.freifunk.net/karte/#/c025e9b8d0a6)
- [minstedt-freifunk-50](https://hannover.freifunk.net/karte/#/3810d5423c31)
- [platz-freifunk99](https://hannover.freifunk.net/karte/#/30b5c2ee799e)