freifunkh/site #56: @AiyionPrime By the way do we have a PGP key for our key submission email address? I guess no. So what...

AiyionPrime, 2021-02-19:
@AiyionPrime By the way do we have a PGP key for our key submission email address? I guess no. So what...

_Originally posted by @CodeFetch in https://github.com/freifunkh/site/issues/54#issuecomment-782287810_

AiyionPrime, 2021-02-19:
- [ ] talk with the other key signers
- [ ] create key pair
- [ ] distribute pubkey to various servers
- [ ] place pubkey on homepage
- [ ] place pubkey or id in frimware image to be shown for the pre reboot key message

AiyionPrime, 2021-02-22:
@1977er @moridius @lemoer @Dark4MD
Wie groß sind euere Abneigungen gegen pgp?
Ich würde anbieten, verschlüsselte key-mails darüber anzunehmen, wäre aber gerne nicht der einzige.

Dark4MD, 2021-02-22:
Hab mich schonmal mit befasst, ging mit Thunderbird Addon relativ easy. Hatte dann aber keinen größeren nutzen und habs liegen gelassen.
Könnte ich dafür aber wohl wieder einrichten. Mein selfhostet Webmailer Rainloop unterstützt das wohl sogar auch.

1977er, 2021-02-22:
Ich habe nichts gegen gpg. Die 1-2x, die man das in 10 Jahren braucht, kann ich gerne jeweils eine alternative Mailsuite installieren, um die Mails dann zu beantworten.

AiyionPrime, 2021-02-23:
Die Gültigkeit habe ich auf zwei Jahre gesetzt, das Passwort verschicke ich gerne verschlüsselt.
Eure Signaturen hätte ich gern.

`gpg --import keys_public.key`
`gpg --allow-secret-key-import --import keys_private.key`

AiyionPrime, 2021-02-23:
I think a good way to share the key would be our website on a https only landing page, that contains info on the key and a link to a proper gpg tutorial?

Other not excluding options would be a pubkey to download from image or key id baked in firmware.

AiyionPrime, 2021-02-23:
The downloadable pubkey is not easy without js, as the router reboots shortly after it delivered the last page.

One could implement the download link as js with preloaded content though.