freifunkh/site #57: Move Schwarmstedt from Umland to own domain

CodeFetch, 2021-02-24:
With more than 300 nodes Umland is already twice as big as our second biggest domain. Schwarmstedt is the biggest mesh in Umland and should therefore be moved to a dedicated domain.

CodeFetch, 2021-02-24:
Hm... When thinking about it, this might confuse users. Instead I propose to add a prefix to the names of the existing domains which reflect districts of Hannover e.g. "Hannover: Misburg" and then split the Umland domain further into regions of the region of Hannover like in this map with the prefix like e.g. "Region: Neustadt", "Region: Springe": https://www.hannover.de/Leben-in-der-Region-Hannover/Gesundheit/Gesundheitsschutz/Coronavirus-in-der-Region-Hannover/Infektionen-mit-COVID-19-in-der-Region-Hannover

1977er, 2021-02-24:
Do we have/need code to move nodes from ie. `Sehnde` to `Umland: Sehnde`?

Dark4MD, 2021-02-24:
For the existing Domains it would just be a display name change in the site and the meshviewer, if we leave the internal names without the prefixes like it is right now.

Dark4MD, 2021-02-24:
Would look something like this (i hope i got all Region and Umland definitions right)
https://github.com/Dark4MD/site/commit/b5b12589f427296a7e5f514261f8d521b86078ad
Btw. we still have legacy as default domain. I'm not sure which would be best but maybe we do something like "Other" or "Umland: Other" as default.

1977er, 2021-02-24:
@Dark4MD The domX files are generated from `gen.py`, aren't they?

CodeFetch, 2021-02-24:
@1977er Yes, using gen.py. Manu is right. Only the Umland nodes will be moved to new domains and the rest are just name changes.

AiyionPrime, 2021-02-25:
> Btw. we still have legacy as default domain. I'm not sure which would be best but maybe we do something like "Other" or "Umland: Other" as default.

I don't think we should put them in an already big domain like umland, but implement a default domain like leetfeld? for this purpose.

Maybe we could even setup some sort of notification on the nodes cli, when the domain is not set properly.

bschelm, 2021-06-03:
Please remember to add the domain-switcher to the next firmware, to push the routers starting with sgs* to the domain Schwarmstedt. There are still too many routers in Schwarmstedt in the domain "Umland".

AiyionPrime, 2023-03-20:
This requires Firmware newer than vH25.

AiyionPrime, 2023-04-25:
and is therefore effectively blocked by #87