freifunkh/site #58: set primary domain properly

AiyionPrime, 2021-03-11:
https://gluon.readthedocs.io/en/latest/features/multidomain.html
strg+f centauri

Der erste Eintrag in der Liste ist die primäre Domain, bei uns ist das derzeit die letzte...

AiyionPrime, 2021-03-12:
This might become a bigger problem, as luas table does not care for ordering at all, iirc.
That gluons lua parser does, is not accoring to standard and a rare exception among other parsers...

I hope their documentation is just outdated...

lemoer, 2021-03-13:
You misinterpreted the documentation here. The the primary domain code is defined by the name of the *.json or *.conf file.

AiyionPrime, 2021-03-13:
That'd be great;
but then this sentence should be removed, I think.
>  -- multiple codes/names can be defined, the first one is the primary name
  -- additional aliases can be defined

The combination of that info and
>  Each domain configuration file is named after its primary domain_code
might imply the first being the source of info and the latter being an optional naming scheme?

But I like the idea of reading the primary domain name from file; that makes stuff easier.

closed 2021-03-13 by AiyionPrime