freifunkh/site #62: Feature Req: additional 10.0.0.1 nextnode address

1977er, 2021-07-21:
This is more or less a work around the bug, that `n.ffh.zone` is not resolvable when the uplink is not present. In that case you need to know the domainnumber in order to access the status page via its IP address:

`fdca:ffee:8:<dom>::1` or `10.<dom>.0.1`

If there were an additional `10.0.0.1`, things would be much easier.

AiyionPrime, 2021-07-24:
> n.ffh.zone is not resolvable when the uplink is not present.

Why is that? I always assumed it was a local dns which responds without centralized support?

lemoer, 2021-07-24:
IIRC that n.ffh.zone does not work without uplink is a bug. And that bug
already exists for some time. I discussed that at congress 2019 with
neoraider.

But he was not very convinced that we should continue to support local dns
names in gluon at all. At that point his opinion was that there is not much
worth in it. But I think he agreed that we could fix this bug. But nobody
did so far.

Actually, I am also unsure what the bug is all about.

>

lemoer, 2021-07-24:
Just adding another IPv4-Address does not really work directly. As DHCP only comes from the gateway, clients wouldn't have neither an IPv4 address nor they would have routing to that address without uplink.

In IPv6, this would be theoretically possible, as the router emits router advertisements. But then we would need to add additional routes (which are not default routes) to the router advertisements. This would require most likely a non-upstreamable gluon patch.

1977er, 2021-08-01:
I agree on IPv4. Unfortunately your might be right with IPv6, too. ;-)

Closing.

closed 2021-08-01 by 1977er