freifunkh/site #79: vH25->vH34 Release-Test

AiyionPrime, 2023-03-10:
@lemoer und ich haben neulich schonmal länger über das Testen von Geräten gesprochen;
bevor wir da eine automatisierte Lösung anstreben und ewig drauf warten, können wir das Prozedere aber ja mal vorher manuell durchgehen und dabei vielleicht erkennen, was wir noch nicht genug bedacht haben.

Wir werden nicht alle Geräte testen können und planen deshalb schadensbegrenzend die TOP 10 oder 20? im Blick zu haben und damit wenigstens die größten Ausfälle zu vermeiden.

Bei der vH25 befinden sich in den TOP 20 unter anderem die ganzen ungemoddeten 841er; ich würde vorschlagen, wir listen hier mal die TOP zehn der vH25 Geräte, die noch Gluon-Support haben.

Das wäre dann diese Liste:

- AVM FRITZ!Box 4040
- AVM FRITZ!Box 4020
- TP-LINK Archer C7 v5
- TP-Link TL-WR1043N/ND v4
- ~~TP-Link TL-WR841N/ND v11~~
- ~~TP-Link TL-WR841N/ND v9~~
- TP-Link TL-WR1043N/ND v3
- TP-Link CPE210 v1.1
- ~~TP-Link TL-WR841N/ND v10~~
- Netgear R6120
- ~~TP-Link TL-WR940N v6~~
- AVM FRITZ!Box 7360 SL
- TP-Link Archer C7 v2
- TP-LINK TL-WR1043N v5

Natürlich ist jeder Test willkommen; vielleicht können wir die Ergebnisse und ihre Tester hier erstmal sammeln und dann gucken wir, ob uns die Top zehn schon genügen, ob wir lieber die Top 20 testen wollen, oder ob es aus anderen Communities schon Verdachtsmomente für andere Geräte gibt, die problematisch sein könnten.

Schreibt gerne einfach drunter, wenn ihr ein Update für eines der Modelle getestet habt.

@lemoer fangen wir an mit:
- der Router kommt nach dem Update von alleine wieder hoch
- er läuft drei Tage stabil
- W-LAN funktioniert noch auf allen Bändern
  - Client
  - Mesh
- Mesh-VPN klappt
- Alle Ports funktionieren noch wie vorher (client ist wo client war, uplink ist wo uplink war)

Der zu testende Pfad ist aktuell von vH25 auf vH34 und dauert damit etwa drei Tage.

Jemand weitere Ideen und oder Einwände? @freifunkh/all

AiyionPrime, 2023-03-12:
Eine CPE 210 habe ich hier nicht; lediglich eine 510.

Dafür aber einen 1043v3.

Habe den mal auf vH25 frisch augfgesetzt, ihm einen fastd-key spendiert und seinen autoupdate-branch auf beta gestellt.

Innerhalb der nächsten Stunde wird er sich updaten (das mache ich zu gunsten des Tests nicht von Hand) und danach kann ich testen, was geht und was nicht.

http://[2a02:790:ff:1014:ee08:6bff:fe8a:d210]/cgi-bin/status

AiyionPrime, 2023-03-12:
Sieht gut aus, werde Dienstag festhalten, wie das Ergebnis aussieht.

Wenn ihr auch einen Router gefunden habt, mit dem ihr testen wollt, schreibt das gerne hier drunter, damit wir alle Geräte erwischen und nicht unnötig doppeln.

AiyionPrime, 2023-03-13:
Denkt dran, dass es wichtig ist, von der vH25 und nicht von irgendeiner Firmware zu starten :)

AiyionPrime, 2023-03-13:
> Ich hab einen C7 v5 da, eine 1043 v3 und v4, eine CPE210 V1.
aber ich muss ja nicht doppelt testen. ICh mach dann erst mal den C7

_@bschelm in hackint#freifunkh_

Ich hab hier noch nen 7360. Den kann ich ab morgen Abend testen.
_@1977er in hackint#freifunkh_

AiyionPrime, 2023-03-13:
Habe @bschelm's CPE210 aufgegabelt und auf die vH25 geflasht wie den 1043er.
Ergebnisse dann in drei Tagen.

Dark4MD, 2023-03-14:
Moin,
ich teste den TL-WR1043v4 und die AVM 4020
Hab auch noch nen C7v2 der eigentlich in betrieb ist. Eventuell tausch ich den kurzfristig aus zum testen falls kein anderer den frei hat.

Dark4MD, 2023-03-15:
Die 7360 die ich habe ist definitiv ne v1 und nun auch am laufen.

AiyionPrime, 2023-03-15:
Der R6120 ist jetzt bei mir zuhause; ich muss später mal gucken, an welchen Injektor ich den gefahrlos hängen kann.

Dark4MD, 2023-03-20:
Ergebnisse gibts im laufe der Woche, grundsätzlich laufen die Geräte aber und zumindest Wifi Mesh und VPN ist stabil. Hab jetzt meinen C7v2 noch frei gemacht und lass den auch laufen. Uptime bei der 4020 und 1043v4 sind aktuell so mies weil die steckdose an der die Hängen aussetzer hat (hab da ne Wifi Steckdose zwischen zum abschalten für meine experimente, scheint aber wackelig geworden zu sein).

AiyionPrime, 2023-03-24:
vH34 wurde gestern für alle Geräte ausser der EAP225v1 veröffentlicht.
Der Test ist soweit abgeschlossen. Vielen Dank allerseits!

closed 2023-03-24 by AiyionPrime