freifunkh/site #84: vH25 im Kernteam loswerden

AiyionPrime, 2023-04-25:
Moin zusammen. Ich schrieb gerade die größten Problembetreiber an; also Leute die beharrlich auf vH25 geblieben sind.

Das ganze würde besser aussehen, wenn wir als Kernteam nicht effektiv der größte Problembetreiber wären.
Ich habe hier mal unsere Problemfälle gelistet; die vH25 Baustelle wäre ohne uns also erheblich kleiner.

Ich würde mich freuen, wenn ihr euch darum bemüht, die entsprechenden Geräte zeitnah zu aktualisieren/zu tauschen, damit wir vH25 loswerden können.

Sobald ihr eure Geräte erledigt habt, könnt ihr euch hier gerne wieder austragen.

Leon
- [ ] https://hannover.freifunk.net/karte/#/de/map/18a6f7ba79b8 (Jakob)
- [ ] https://hannover.freifunk.net/karte/#/de/map/6872514c4ef0 (springe.freifunk.net_PicoStation)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327cefc54 (springe.freifunk.net_Indoor)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086baae80c (Ratskeller_FW-01)
- [ ] https://hannover.freifunk.net/karte/#/de/map/18a6f73ddb74 (FFS_Museum-Burghof)
- [ ] https://hannover.freifunk.net/karte/#/de/map/00037fbef0bd (springe.freifunk.net_Kirche-Nord-5GHz)
- [ ] https://hannover.freifunk.net/karte/#/de/map/f4f26dd5f6be (Ratskeller Büro 1)
- [ ] https://hannover.freifunk.net/karte/#/de/map/e063da6dfd53 (FFS-B Freibad Bennigsen)
- [ ] https://hannover.freifunk.net/karte/#/de/map/98ded0318c6c (Bähre-II)
- [ ] https://hannover.freifunk.net/karte/#/de/map/98ded03178fa (Bähre-III)
- [ ] https://hannover.freifunk.net/karte/#/de/map/18a6f73de58e (FFS_Kirche-Ost)
- [ ] https://hannover.freifunk.net/karte/#/de/map/18a6f73de4c8 (FFS_Museum-West)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086baadb48 (Ratskeller_Restaurant)

@bschelm 
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327c70272 (FFH-FWH-MunzelerStr-1OG-Zimmer)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327e719bc (WBH-1)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e32781ecb6 (Vulkan-Sauna-Kueche)
- [ ] https://hannover.freifunk.net/karte/#/de/map/c04a00e3b8ba (WBH-2)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e32781ed98 (Vulkan-Sauna-Gruen)
- [ ] https://hannover.freifunk.net/karte/#/de/map/6470025c2548 (FFH-FWH-Kopernikusstr-W2)
- [ ] https://hannover.freifunk.net/karte/#/de/map/10feedb66064 (FFH-Asternstr-13)
- [ ] https://hannover.freifunk.net/karte/#/de/map/e8de27c86a2c (WBH-3)
- [ ] https://hannover.freifunk.net/karte/#/de/map/f4f26dff3b6c (Liebhabereien-2)
- [ ] https://hannover.freifunk.net/karte/#/de/map/8416f9c8b550 (NDS-Die-PARTEI)
- [ ] https://hannover.freifunk.net/karte/#/de/map/10feedc45fa2 (FFH-FWH-Hebbelstr-1)

@Dark4MD 

- [ ] https://hannover.freifunk.net/karte/#/de/map/18a6f7d977d6 (NRUE WiFoe Indoor 120)
- [ ] https://hannover.freifunk.net/karte/#/de/map/98ded088860a (NDS-NRUE WiFoe Outdoor 217)
- [ ] https://hannover.freifunk.net/karte/#/de/map/98ded088860e (NRUE WiFoe Outdoor 207)
- [ ] https://hannover.freifunk.net/karte/#/de/map/18a6f7d96326 (NRUE WiFoe Indoor 119)
- [ ] https://hannover.freifunk.net/karte/#/de/map/687251661453 (NRUE WiFoe Outdoor 213)
- [ ] https://hannover.freifunk.net/karte/#/de/map/647002b5e860 (Manu.WTF-Hof-Stall)
- [ ] https://hannover.freifunk.net/karte/#/de/map/e894f63f3bf8 (Manu.WTF-Hof-Werkstatt)
- [ ] https://hannover.freifunk.net/karte/#/de/map/788a20f31cbd (Manu.WTF-Hof-Garage)
- [ ] https://hannover.freifunk.net/karte/#/de/map/8416f9c89c54 (Manu.WTF-Hof-Büro)
- [ ] https://hannover.freifunk.net/karte/#/de/map/74acb920cad7 (NDS-Manu.WTF-Mauer)

@Manawyrm?

- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327bd993a (Alfeld-Seminarstrasse)
- [ ] https://hannover.freifunk.net/karte/#/de/map/8416f9af8910 (Alfeld-Winzenburger-Strasse)

@tobby88 

- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086b7cda3a (freifunk-am-egalsten)
- [ ] https://hannover.freifunk.net/karte/#/de/map/c46e1fe7ce34 (freifunk-egal)

Manawyrm, 2023-04-25:
> Ich würde mich freuen, wenn ihr euch darum bemüht, die entsprechenden Geräte zeitnah zu aktualisieren/zu tauschen, damit wir vH25 loswerden können.

Hab nicht ordentlich mitgelesen -- alle 841er sind hier das Problem, richtig?

AiyionPrime, 2023-04-25:
@Manawyrm Alles was dir oder deinen Liebsten gehört und auf der Karte rote Banner hat.
Die beiden Router, die dich betreffen, habe ich oben unter deinem Namen markiert;
kann natürlich sein, dass mir da welche entgangen sind. Deine bdien sind 841er, aber es gibt auch andere nicht mehr unterstützte Modelle.
Und es gibt auch Leute, die einfach nur updaten bräuchten.

Wenn dir mehr einfallen, trag sie gerne nach.

tobby88, 2023-04-26:
Zumindest die beiden bei mir genannten Router (ich meine, es müssten eigentlich sogar drei sein, aber evtl. ist einer tot) werden weiterhin benötigt. Haben wir noch Ersatz rumfliegen? Z.B. aufgerüstete 841er, die ich dafür nehmen könnte?

AiyionPrime, 2023-04-26:
Ersatzlos abachaffen ar nie mien Ziel :)
@bschelm berichtete nelich von noch neun gemoddeten 841ern.

Eventuell ist die Anschaffung von drei DAP-x1860 aber zielführender für dich.
Kosten weniger als 20€, haben Wifi 6, Dualband und werden länger unterstützt werden, als die 841er.

tobby88, 2023-04-26:
Dann frag ich ihn Mal.
Das sind nicht meine Geräte und ich wüsste nicht warum ich die dann kaufen sollte 😀

AiyionPrime, 2023-04-26:
In dem Fall empfehle ich dir, was ich auch Bernd empfehle. Der schreibt auch auf jeden Router den er trifft seine Mail-Adresse :P