freifunkh/site #89: Schneerens Heimrouter ersetzen lassen

AiyionPrime, 2023-04-25:
Frau L. wurde angeschrieben, mal sehen, ob und wann da Antwort kommt.

- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086b692144 (Residenz_Meerblick_05)
- [ ] https://hannover.freifunk.net/karte/#/de/map/f4f26df2a132  (freifunk-Schneeren-hbs10)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086bab4afe (ResidenzMeerblick07)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327cf0ce4 (freifunk-Schneeren-HBS04)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086bab6f9e (Residenz_Meerblick_02)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327cf0b44 (freifunk-Schneeren-HBS02)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086bab4854 (freifunk-Schneeren-HBS07)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327cf0d28 (freifunk-Schneeren-HBS05)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086b6f8578 (freifunk-Schneeren-HBS09)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086b69213e (Residenz_Meerblick_01)
- [ ] https://hannover.freifunk.net/karte/#/de/map/60e327ee146a (freifunk-SchneerenHBS09)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086b7cdb5c (freifunk-Schneeren-HBS01)
- [ ] https://hannover.freifunk.net/karte/#/de/map/ec086b6132da (Residenz_Meerblick_03)

1977er, 2025-08-30:
# Aktuelle Situation:

## Residenz
* [Residenz_Meerblick_01](https://hannover.freifunk.net/karte/#!/de/map/ec086b69213e) WR841N/ND v11
* [Residenz_Meerblick_02](https://hannover.freifunk.net/karte/#!/de/map/ec086bab6f9e) WR841N/ND v11
* [Residenz_Meerblick_03](https://hannover.freifunk.net/karte/#!/de/map/ec086b6132da) WR841N/ND v11
* [Residenz_Meerblick_04](https://hannover.freifunk.net/karte/#!/de/map/ec086b61334a) WR841N/ND v11
* [ResidenzMeerblick07](https://hannover.freifunk.net/karte/#!/de/map/ec086bab4afe) WR841N/ND v11

## HBS
* [freifunk-Schneeren-HBS01](https://https://hannover.freifunk.net/karte/#!/en/map/ec086b7cdb5c) WR841N/ND v11
* [freifunk-Schneeren-HBS07](https://hannover.freifunk.net/karte/#!/en/map/ec086bab4854) WR841N/ND v11
* [freifunk-Schneeren-hbs10](https://hannover.freifunk.net/karte/#!/en/map/f4f26df2a132) WR841N/ND v11
* [freifunk-Schneeren8](https://hannover.freifunk.net/karte/#!/en/map/60e327cee3ea) WR841N/ND v10

## Godehard
* [schneeren n11](https://hannover.freifunk.net/karte/#!/en/map/30b5c2d9a6c0) WR841N/ND Mod (16M) v9
* [schneeren n10](https://hannover.freifunk.net/karte/#!/de/map/f4f26ddc9930) WR841N/ND Mod (ATH79 16M) v10
* [schneeren n1](https://hannover.freifunk.net/karte/#!/de/map/60e3279aaf68) WR841N/ND Mod (ATH79 16M) v10
* [schneeren n6](https://hannover.freifunk.net/karte/#!/de/map/60e327cf1762) WR841N/ND Mod (ATH79 16M) v10
* [schneeren n4](https://hannover.freifunk.net/karte/#!/de/map/ec086b353bf8) WR841N/ND Mod (ATH79 16M) v11
* [DGH Schneeren](https://hannover.freifunk.net/karte/#!/de/map/60e327ee0e26) WR841N/ND Mod (ATH79 16M) v10

1977er, 2025-09-02:
HBS und Residenz werden zurückgebaut, Geräte dann abholbar. Wv 03.11.2025 gesetzt.