freifunkh/site #91: EAP225v3 upstreamen

AiyionPrime, 2023-04-25:
Wir könnten mal zusehen, dass wir den Patch aus https://github.com/freifunkh/site/commit/f528dae16a03debbf4c4a94ceb291d23df0806a1 wieder loswerden.

https://github.com/freifunkh/site/blob/master-wireguard/patches/0007-ath79-generic-add-support-for-TP-Link-EAP-225-Outdoo.patch

Da wäre ein OpenWrt Backport notwendig, der nur Chance auf Erfolg hat, wenn das Gerät in Gluon sauber tut.

@s-2 Ich habe im Kopf, dass da was unsauber lief, hatte sich das erledigt?

s-2, 2023-04-25:
> Ich habe im Kopf, dass da was unsauber lief, hatte sich das erledigt?

Sollte erledigt sein sobald https://github.com/freifunk-gluon/gluon/pull/2879 gemerged ist.

AiyionPrime, 2023-04-25:
Dann können wir das Donerstag direkt testen, wenn wir mit #88 durch sind.

AiyionPrime, 2023-04-27:
Router läuft auf der aktuellen nightly scheinbar gut. @s-2 könnte den PR die Tage mal bei gluon einreichen.

s-2, 2023-04-28:
Wir würden also den kompletten Geräte-Support (inkl. .dts / Image-Makefile, aber ohne factory image) in gluon patchen?
Für einen backport in OpenWrt 22.03 müsste dort auch `firmware-utils` hochgezogen werden, da die IDs für das Gerät im `tplink-safeloader` hardcoded sind (deshalb wird bei uns aktuell nur das sysupgrade generiert).

AiyionPrime, 2023-04-28:
Je nachdem, wie machbar das hochziehen der firmware-utils an der Stelle ist, mach einen Backprt PR bei OpenWrt auf, der das enthält, oder nicht.