freifunkh/site #100: wg Tunnelblockade

1977er, 2023-06-03:
### Symptome:
- Pakete, die auf dem Supernode ins wg Interface gehen, kommen auf der Knoten am mesh_wg Interface nicht mehr an. (tcpdump -ni mesh_wg inbound)
- Pakete sind auf br-wan noch sichtbar.
- wg Handshakes werden erneuert.
- CPU Load senkt sich deutlich ab (auf <20% beim ER-X).
- Speicherauslastung steigt um einen kleinen, aber stabilen Sockelbetrag an.

### Resultierende Symptome:
- Batman VPN Neighbours geht auf 0 runter.
- TX Link Quality to Supernodes (aka. TQ) bekommt keine Telemetrie mehr.
- RX Link Quality from Supernodes (aka. reversed TQ) geht auf <5% runter.
- Forward TX Traffic sinkt auf 0.

### Siehe aus:
- Mailthread: "Ausfall-Check"
- Grafana: https://stats.ffh.zone/d/EGysRPsMz/router-fur-meshviewer?var-node=74acb9a77737&from=1685753778880&to=1685786777470&orgId=1
- debug_info: https://gist.github.com/1977er/7ed4ba1028b903a8957be8bdf61beecb
- debug_info: https://gist.github.com/1977er/3c52bf5ffba9273ef54766f286a87c6f

### Bisherige Maßnahmen:
- Monitoring checkt, ob vpn_neighbours=0 (funktioniert nur bei Sites mit mehreren Uplinks)

lemoer, 2023-06-22:
Debug output from today:

## wg show on UFU-FWH-E106-Woermannstr-Technik1

```
root@UFU-FWH-E106-Woermannstr-Technik1:/tmp# wg show
interface: wg_mesh
  public key: lp7KvqqBam9boZyfJJgLIjd8jWF7nREqGA53WW5pfSE=
  private key: (hidden)
  listening port: 52231
  fwmark: 0x1

peer: SN10jGGoMekUFtCenlM1RMbnFh3fqJnhjTXnpNWqJ1A=
  endpoint: 81.3.6.91:51816
  allowed ips: fe80::1/128
  latest handshake: 2 seconds ago
  transfer: 1.75 TiB received, 192.12 GiB sent
```

## Outbound on UFU-FWH-E106-Woermannstr-Technik1 wireguard interface

```
root@UFU-FWH-E106-Woermannstr-Technik1:/tmp# tcpdump -n -i wg_mesh -c 5 outbound 
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on wg_mesh, link-type RAW (Raw IP), capture size 262144 bytes
21:02:22.415229 IP6 fe80::277:2bff:fe2a:2f28.58294 > fe80::1.4789: VXLAN, flags [I] (0x08), vni 16317534
02:3e:b3:3a:4a:fa > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x4305), length 162: 
	0x0000:  000f 2c00 11d1 0c62 d65e 5062 8583 be18  ..,....b.^Pb....
	0x0010:  0305 c0b0 0093 0038 0401 0028 013e 0003  .......8...(.>..
	0x0020:  a145 edb7 8001 0000 e879 9090 8000 0000  .E.......y......
	0x0030:  618f fa0b 0000 0000 1300 0000 3007 4de1  a...........0.M.
	0x0040:  3f73 0000 0602 0004 0100 0000 0201 0000  ?s..............
	0x0050:  000f 2a00 fe08 c294 a6ef 7a86 2ed3 be18  ..*.......z.....
	0x0060:  0305 c0b0 0068 002c 0401 001c 0197 0003  .....h.,........
	0x0070:  041e 320d 8001 0000 4d22 4f2a 8000 0000  ..2.....M"O*....
	0x0080:  d6e4 041b 0000 0000 0602 0004 0100 0000  ................
	0x0090:  0201 0000                                ....
21:02:22.425599 IP6 fe80::277:2bff:fe2a:2f28.58294 > fe80::1.4789: VXLAN, flags [I] (0x08), vni 16317534
02:3e:b3:3a:4a:fa > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x4305), length 150: 
	0x0000:  000f 3100 ba50 2c2a be18 0305 c0b3 be18  ..1..P,*........
	0x0010:  0305 c0b0 00f0 002c 0401 001c 0141 0003  .......,.....A..
	0x0020:  b244 398b 8001 0000 fb78 44ac 8000 0000  .D9......xD.....
	0x0030:  7e27 3918 0000 0000 0602 0004 0100 0000  ~'9.............
	0x0040:  0201 0000 000f 2b00 e9ad 07e2 be2f 2c4b  ......+....../,K
	0x0050:  bbc3 be18 0305 c0b0 0081 002c 0401 001c  ...........,....
	0x0060:  016b 0003 65ad 73c8 8001 0000 2c91 0eef  .k..e.s.....,...
	0x0070:  8000 0000 0944 f095 0000 0000 0602 0004  .....D..........
	0x0080:  0100 0000 0201 0000                      ........
21:02:22.535510 IP6 fe80::277:2bff:fe2a:2f28.58294 > fe80::1.4789: VXLAN, flags [I] (0x08), vni 16317534
02:3e:b3:3a:4a:fa > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x4305), length 218: 
	0x0000:  000f 2c00 f565 8fdf 5e87 400e b7d3 be18  ..,..e..^.@.....
	0x0010:  0305 c0b0 0089 002c 0401 001c 01ba 0003  .......,........
	0x0020:  d01b 8003 8001 0000 9927 fd24 8000 0000  .........'.$....
	0x0030:  568f 2ea2 0000 0000 0602 0004 0100 0000  V...............
	0x0040:  0201 0000 000f 2c00 5386 3a55 4e27 fe8a  ......,.S.:UN'..
	0x0050:  718b be18 0305 c0b0 0089 002c 0401 001c  q..........,....
	0x0060:  0128 0003 37ae 48e3 8001 0000 7e92 35c4  .(..7.H.....~.5.
	0x0070:  8000 0000 b682 98ae 0000 0000 0602 0004  ................
	0x0080:  0100 0000 0201 0000 000f 2b00 8021 66f5  ..........+..!f.
	0x0090:  8253 adf4 2fab be18 0305 c0b0 0070 002c  .S../........p.,
	0x00a0:  0401 001c 01a6 0003 2dd5 ff67 8001 0000  ........-..g....
	0x00b0:  64e9 8240 8000 0000 ccd7 5534 0000 0000  d..@......U4....
	0x00c0:  0602 0004 0100 0000 0201 0000            ............
21:02:22.635524 IP6 fe80::277:2bff:fe2a:2f28.58294 > fe80::1.4789: VXLAN, flags [I] (0x08), vni 16317534
02:3e:b3:3a:4a:fa > ff:ff:ff:ff:ff:ff, ethertype Unknown (0x4305), length 218: 
	0x0000:  000f 2b00 4dff 2907 3e39 c135 b7fb be18  ..+.M.).>9.5....
	0x0010:  0305 c0b0 0070 002c 0401 001c 013f 0003  .....p.,.....?..
	0x0020:  7dba 5544 8001 0000 3486 2863 8000 0000  }.UD....4.(c....
	0x0030:  1d10 2714 0000 0000 0602 0004 0100 0000  ..'.............
	0x0040:  0201 0000 000f 2c00 7854 616e 323d f939  ......,.xTan2=.9
[... 129 lines cut ...]
	0x0100:  27db 26dc 00c6 002c 0401 001c 01bb 0003  '.&....,........
	0x0110:  a669 7925 8001 0000 ef55 0402 8000 0000  .iy%.....U......
	0x0120:  f34d db32 0000 0000 0602 0004 0100 0000  .M.2............
	0x0130:  0201 0000 000f 3000 8230 55ee 5e03 c2cf  ......0..0U.^...
	0x0140:  9923 2aab 8a15 cdcd 00c3 002c 0401 001c  .#*........,....
	0x0150:  0125 0003 3192 e5ae 8001 0000 78ae 9889  .%..1.......x...
	0x0160:  8000 0000 3f6b 668e 0000 0000 0602 0004  ....?kf.........
	0x0170:  0100 0000 0201 0000                      ........
5 packets captured
6 packets received by filter
0 packets dropped by kernel
```

# sn10

## wg show on sn10

``` 
[root@sn10]:~ # wg show  wg-16 | grep lp7KvqqBam9boZyfJJgLIjd8jWF7nREqGA53WW5pfSE= -A 5
peer: lp7KvqqBam9boZyfJJgLIjd8jWF7nREqGA53WW5pfSE=
  endpoint: 89.183.250.69:52231
  allowed ips: fe80::277:2bff:fe2a:2f28/128
  latest handshake: 10 seconds ago
  transfer: 188.98 GiB received, 1.77 TiB sent
```

lemoer, 2023-06-22:
It also doesn't seem to be related to the vxlan traffic that we are sending into the wg interface. If we do a normal ping, we can also just see the handshakes on br-wan:

```
root@UFU-FWH-E106-Woermannstr-Technik1:/tmp# ping fe80::1%wg_mesh
PING fe80::1%wg_mesh (fe80::1%16): 56 data bytes

```

```
root@UFU-FWH-E106-Woermannstr-Technik1:~# tcpdump -n -i wg_mesh outbound and icmp6 -v
tcpdump: listening on wg_mesh, link-type RAW (Raw IP), capture size 262144 bytes
21:41:55.055297 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 161
21:41:56.065291 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 162
21:41:57.075316 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 163
21:41:58.085348 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 164
21:41:59.095278 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 165
21:42:00.105287 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 166
21:42:01.115318 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 167
21:42:02.125331 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 168
21:42:03.135295 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 169
21:42:04.145293 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 170
21:42:05.155292 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 171
21:42:06.165324 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 172
21:42:07.175321 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 173
21:42:08.185316 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 174
21:42:09.195293 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 175
21:42:10.205302 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 176
21:42:11.215292 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 177
21:42:12.225289 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 178
21:42:13.235292 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 179
21:42:14.245295 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 180
21:42:15.255438 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 181
21:42:16.265321 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 182
21:42:17.275321 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 183
21:42:18.285300 IP6 (flowlabel 0xd5ca2, hlim 64, next-header ICMPv6 (58) payload length: 64) fe80::277:2bff:fe2a:2f28 > fe80::1: [icmp6 sum ok] ICMP6, echo request, seq 184
```

```
root@UFU-FWH-E106-Woermannstr-Technik1:~# tcpdump -n -i br-wan outbound and host 81.3.6.91
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on br-wan, link-type EN10MB (Ethernet), capture size 262144 bytes
21:41:55.388648 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
21:42:10.527058 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
```

lemoer, 2023-06-22:
```
root@UFU-FWH-E106-Woermannstr-Technik1:~# wg showconf wg_mesh
[Interface]
ListenPort = 52231
FwMark = 0x1
PrivateKey = uKtTbY0wVTwMLXH4j/34YJi1V4ZkDYxFh3+KdttVCUc=

[Peer]
PublicKey = SN10jGGoMekUFtCenlM1RMbnFh3fqJnhjTXnpNWqJ1A=
AllowedIPs = fe80::1/128
Endpoint = 81.3.6.91:51816
```

lemoer, 2023-06-22:
Also, the TX counter of wireguard shows that packets are not sent out (second last column): 

```
root@UFU-FWH-E106-Woermannstr-Technik1:~# wg show all dump; sleep 15; wg show all dump
wg_mesh	uKtTbY0wVTwMLXH4j/34YJi1V4ZkDYxFh3+KdttVCUc=	lp7KvqqBam9boZyfJJgLIjd8jWF7nREqGA53WW5pfSE=	52231	0x1
wg_mesh	SN10jGGoMekUFtCenlM1RMbnFh3fqJnhjTXnpNWqJ1A=	(none)	81.3.6.91:51816	fe80::1/128	1687464100	1920192174532	206288920592	off
wg_mesh	uKtTbY0wVTwMLXH4j/34YJi1V4ZkDYxFh3+KdttVCUc=	lp7KvqqBam9boZyfJJgLIjd8jWF7nREqGA53WW5pfSE=	52231	0x1
wg_mesh	SN10jGGoMekUFtCenlM1RMbnFh3fqJnhjTXnpNWqJ1A=	(none)	81.3.6.91:51816	fe80::1/128	1687464115	1920192233560	206288920684	off
```

Only 92 bytes are seen in 15 seconds only. This is excactly the size of one handshake packet.

lemoer, 2023-06-22:
Also an inbound trace on UFU-WFH-E106-Technik1 on br-wan:
```
root@UFU-FWH-E106-Woermannstr-Technik1:~# tcpdump -n -i any host 81.3.6.91 and inbound
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
22:12:53.556666 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 320
22:12:53.556714 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 320
22:12:53.592908 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:53.592959 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:53.651267 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 320
22:12:53.651317 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 320
22:12:53.723720 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:53.723768 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:53.723923 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 320
22:12:53.723942 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 320
22:12:53.753076 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:53.753132 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:53.856151 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 384
22:12:53.856198 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 384
22:12:53.856321 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:53.856346 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:53.951600 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 128
22:12:53.951648 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 128
22:12:54.042231 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 192
22:12:54.042275 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 192
22:12:54.042389 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:54.042404 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:54.042455 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:54.042478 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:54.059258 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 240
22:12:54.059304 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 240
22:12:54.204977 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.205059 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.335885 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 240
22:12:54.335933 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 240
22:12:54.340871 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.340925 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.519126 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 128
22:12:54.519190 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 128
22:12:54.519374 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 256
22:12:54.519394 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 256
22:12:54.519446 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 224
22:12:54.519461 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 224
22:12:54.538861 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.538910 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.580684 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.580730 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 176
22:12:54.670918 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
22:12:54.670967 IP 81.3.6.91.51816 > 192.168.178.32.52231: UDP, length 160
^C
44 packets captured
44 packets received by filter
0 packets dropped by kernel
```

lemoer, 2023-06-22:
I just found this in dmesg:
````
[373474.182938] ------------[ cut here ]------------
[373474.192353] WARNING: CPU: 0 PID: 0 at kernel/rcu/tree.c:624 rcu_eqs_enter.constprop.0+0xd8/0xe0
[373474.209825] Modules linked in: iptable_nat batman_adv xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT wireguard nf_nat nf_conntrack libchacha20poly1305 ipt_REJECT ebtable_nat ebtable_filter ebtable_broute cfg80211 xt_time xt_tcpudp xt_quota xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_comment xt_addrtype xt_TCPMSS xt_LOG ts_kmp ts_fsm ts_bm poly1305_mips nf_reject_ipv4 nf_log_ipv6 nf_log_ipv4 nf_log_common nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 compat chacha_mips sch_teql sch_sfq sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_pedit act_csum libcrc32c sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow act_skbedit act_mirred act_gact ip6table_mangle ip6table_filter ip6_tables
[373474.210230]  ip6t_REJECT x_tables nf_reject_ipv6 dummy ip_tunnel veth vxlan udp_tunnel ip6_udp_tunnel kpp leds_gpio cls_basic sch_tbf sch_ingress gpio_button_hotplug crc32c_generic
[373474.415982] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.10.161 #0
[373474.428266] Stack : 809a0000 00000000 00000000 8007c664 80830000 8072d760 00000000 00000000
[373474.445079]         807b3d24 80980000 806fdc74 807c7298 807c6dc7 00000001 807b3cc8 6f3afccb
[373474.461890]         00000000 00000000 806fdc74 807b3b68 ffffefff 00000000 ffffffea 00000000
[373474.478702]         807b3b74 0000017a 807cb9e0 ffffffff 00000000 00000000 00000000 80700000
[373474.495513]         00000009 3c016ed8 000153ac 00000000 00000000 803a42ec 00000000 80980000
[373474.512324]         ...
[373474.517357] Call Trace:
[373474.522404] [<80007fc0>] show_stack+0x30/0x100
[373474.531422] [<8031f7c8>] dump_stack+0x9c/0xcc
[373474.540272] [<8002fe64>] __warn+0xc0/0x12c
[373474.548587] [<8002ff2c>] warn_slowpath_fmt+0x5c/0xac
[373474.558629] [<80656d6c>] rcu_eqs_enter.constprop.0+0xd8/0xe0
[373474.570058] [<8045dde4>] cpuidle_enter_state+0x410/0x540
[373474.580790] [<8045dfac>] cpuidle_enter+0x84/0xac
[373474.590163] [<8005e860>] do_idle+0x26c/0x31c
[373474.598824] [<8005eb88>] cpu_startup_entry+0x2c/0x34
[373474.608887] [<80848d00>] start_kernel+0x570/0x598
[373474.618408] 
[373474.621532] ---[ end trace 74d97155b98095dd ]---
```

Not sure if this is related, since this happened 15 days ago.

lemoer, 2023-06-22:
Listening on any interface, also just shows the handshakes:

```
root@UFU-FWH-E106-Woermannstr-Technik1:~# tcpdump -ni any host 81.3.6.91 and outbound
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on any, link-type LINUX_SLL (Linux cooked v1), capture size 262144 bytes
22:53:01.899449 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:01.899506 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:01.899528 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:17.252236 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:17.252308 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:17.252336 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:32.634665 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:32.634747 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:32.634779 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:48.233387 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:48.233449 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:53:48.233470 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:54:03.588697 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:54:03.588774 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:54:03.588801 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:54:18.947856 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:54:18.947911 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
22:54:18.947933 IP 192.168.178.32.52231 > 81.3.6.91.51816: UDP, length 92
^C
18 packets captured
18 packets received by filter
0 packets dropped by kernel
```

I suppose every handshake is shown three times, since I have a linux bridge (or so):
```
root@UFU-FWH-E106-Woermannstr-Technik1:~# brctl show
bridge name	bridge id		STP enabled	interfaces
br-mesh_other		7fff.eaf340edfbcc	no		eth3
							eth1
							eth4
							eth2
br-client		7fff.74acb9a77737	no		bat0
							local-port
br-wan		7fff.74acb9a77737	no		eth0
```

1977er, 2025-01-19:
Todays debugging session showed, that the direction of "its not working vs. its working" may be interchanged, too. Today it was not possible to ping fe80::1%wg_mesh. tcpdump showed that the reply packet is withdrawn somewhere on the way back from the supernode.

lemoer, 2025-01-29:
https://pad.leinelab.org/mKN6BNbHRIi5J7GPChw2FQ?view

lemoer, 2025-01-29:
If this is seen the next time on a supernode (where the supernode get's deaf), please execute:

```
echo 'module wireguard +p' | sudo tee /sys/kernel/debug/dynamic_debug/control
```

And capture logs using journalctl...

I just checked on sn10, that this should work.