freifunkh/site #103: vH38

lemoer, 2025-01-18:
Heyho,

let's setup a new (more current) wireguard build. Let's use this issue to track process.

lemoer, 2025-01-18:
I would propose, we use gluon release v2023.2.4 as base.

lemoer, 2025-01-18:
I am currently unsure which site branch to use. There's master-wireguard and stable-wireguard.

Due to the most recent commit on master-wireguard ([2c8789d](https://github.com/freifunkh/site/commit/2c8789d9a0c1ff9e275690650622c228d0ffe6ef)), the patches do not apply. If I revert this patch, the patches apply to v2023.2.4.

lemoer, 2025-01-18:
## Updating the stable-wireguard branch

According to the README.md, we want to make releases from the stable-wireguard branch. Currently, this branch is one commit ahead and 5 commits behind master-wireguard (if we leave off the broken fc04cd7):

```
* fc04cd7 (origin/master-wireguard) patches: refresh patch for WR841 8M and 16M variants
* 55c17b3 Delete 0002-patches-uradvd-change-ip-lifetimes-to-150-300-seconds.patch
* 1fa3283 Adapted to changes in package definition
* 9a0f62d patches: drop EAP225-Outdoor-v3 patch
* dfbcdd7 patches: refresh patches, drop EAP225-Outdoor-v3 OpenWrt patch
* 9401585 nightly_wireguard-only/site.conf: use jenkins ...
| * 0669244 (tag: vH37, origin/stable-wireguard) patches: refresh upon Gluon v2022.1.4
|/  
* 150f503 site.conf: load packages from firmware.ffh.zone

```

Since the branches have diverged, I am currently unsure, what's the best way to update stable-wireguard?
1. merge commit from 55c17b3 into stable-wireguard?
2. rebase master-wireguard on stable-wireguard so stable-wireguard can be fast-forwarded to 55c17b3?

@AiyionPrime Apart from that, I really like the idea of 0669244. To me, this seems to make perfectly sense to track the gluon version the patches should apply to. This should also land in master-wireguard somehow.

AiyionPrime, 2025-01-18:
this would be vH39, right?
vH38 would be the next fastd firmware.

lemoer, 2025-01-18:
Do we need another fastd release?

Apart from this, I dislike the idea that vH38 could be "a newer release"
than e.g. vH43, if we build the next fastd release in two years after the
wireguard release vH43 was already released. What do you think?

2025, 13:39:

1977er, 2025-01-18:
`They always come in pairs.`

Wir sollten weiterhin immmer zwei parallele Firmwares produzieren, um genau dieses Divergenzproblem zu umgehen.

lemoer, 2025-01-18:
I asked around in the gluon irc if there is a reason to build from v2023.2.x instead of v2023.2.3. 

The answer was:

There is https://github.com/freifunk-gluon/gluon/commit/6f289cf0898e79158f093655be84e549999db217. It fixes a respondd crash on some architectures (at least ARM64).

lemoer, 2025-01-18:
What is done:
- I looked through the image-customizations.lua from @Dark4MD and adjusted it to be a bit more hannover specific. E.g. in contrast to ffmuc, we only want usb packages in x86.
- Aiyion updated the uradvd patches, so we don't have to drop them anymore.
- The drop EAP patch is also on stable-wireguard now (since the device is upstream).
- The mod 841 patch is now rebased to work on v2023.2.x.
- We decided to build v2023.2.4 and add https://github.com/freifunk-gluon/gluon/commit/6f289cf0898e79158f093655be84e549999db217 as a patch. Aiyion is currently doing this.

AiyionPrime, 2025-01-19:
The libgen patch has been added in https://github.com/freifunkh/site/commit/26dac4a770b6411851bd4a820bdb679b642dfa0d

lemoer, 2025-01-20:
# TODO-Liste

- [ ] Migration von fastd-keys nach Wireguard prüfen.

lemoer, 2025-01-21:
Currently, the ipq target does not seem to have a migration path:

(from vH34:)
```
root@...:/tmp# sysupgrade gluon-ffh-vH39.pre-avm-fritz-box-4040-sysupgrade.bin
Tue Jan 21 23:31:33 CET 2025 upgrade: The device is supported, but the config is incompatible to the new image (1.0->1.1). Please upgrade without keeping config (sysupgrade -n).
Tue Jan 21 23:31:33 CET 2025 upgrade: Config cannot be migrated from swconfig to DSA
Image check failed.
```

This issue is most likely related: https://github.com/freifunk-gluon/gluon/issues/2666

AiyionPrime, 2025-01-22:
We could drop the check and evaluate whats missing.
Gluon rebuilds large parts of the network config,
would be interesting to see which are amiss and how hard the info is to gather.

Knowing blocktrron though: For him this is a theoretically trivial matter.
I assume there's something more problematic going on, like ipq40xx DSA not working properly yet.
I'd suspect such a thing to be the reason, why he hasn't tackled this, yet.

1977er, 2025-01-25:
Netgear EX6150v2:
`opkg update` failed due to wrong url:
/etc/opkg/gluon.conf has to point to `http://firmware.ffh.zone/vH39.pre/packages/gluon-ffh-vH39.pre/ipq40xx/generic` instead of  `http://firmware.ffh.zone/packages/gluon-ffh-vH39.pre/ipq40xx/generic`.

AiyionPrime, 2025-01-26:
- [ ] primary mac auf joy-it router prüfen.

lemoer, 2025-02-01:
⚠️ Potential problem

Currently, I tried to build the pre-release vH39~3. However, now the ramips-m7621 target is missing.

The build failes with:
```
#
# configuration written to .config
#
make[1]: Leaving directory '/root/aiyion-vh39/gluon-stable-wireguard/openwrt'
Configuration failed:
 * unable to enable package 'ffda-network-setup-mode'
make: *** [Makefile:190: config] Error 1
```

The issue seem to be that we are trying to include this package from our site:
https://github.com/freifunkh/site/blob/26dac4a770b6411851bd4a820bdb679b642dfa0d/image-customization.lua#L147-L153

AiyionPrime, 2025-08-27:
vH39 has been released on 10th of February this year.
https://hannover.freifunk.net/wiki/Freifunk/FirmwareReleases

closed 2025-08-27 by AiyionPrime