freifunkh/site #104: introduce sqm / cake

1977er, 2025-01-21:
Other communities have had good experiences with sqm/cake in order to retain a bit of fluidity when uplinks are overloaded.
This is currently only supported for fastd. However, an adaptation to wireguard looks feasible at first glance.

Implementation for fastd: https://github.com/freifunk-gluon/gluon/blob/ad321f50e6c2568d7702cdd5b3253b09c0535198/package/gluon-mesh-vpn-fastd/luasrc/usr/lib/lua/gluon/mesh-vpn/provider/fastd.lua

Needs to be moved to wireguard: https://github.com/freifunk-gluon/gluon/blob/ad321f50e6c2568d7702cdd5b3253b09c0535198/package/gluon-mesh-vpn-wireguard/luasrc/usr/lib/lua/gluon/mesh-vpn/provider/wireguard.lua

See also: https://github.com/freifunk-gluon/gluon/commit/2afdfd60f9f71165ae1d17d16936997275772cf4#diff-9ed3603b33816ceecac26a49017e86b4cdef93e36e9a49357315b4279bc018a5R53

lemoer, 2025-01-21:
We can build a firmware to test this. How do we test it then if it is working?

1977er, 2025-01-21:
Either test it live in a FWH, or have a Netflix party at my house which is conntected with 50Mbps only. ;-)

AiyionPrime, 2025-01-22:
Last time I worked in that file it wasn't clear, whether traffic shaping worked at all in our WireGuard setup.